pymeasure/pymeasure #1015: Bug: Exception if loading result file with an empty parameter

poje42, 2024-01-02:
Steps to Reproduce:
* Run the gui.py example
* Clear the "Random Seed" field
* Run the test
* Remove graph
* Load the generated file
* Exception in results.py

An empty string parameter will be saved as `#<Tab>Random Seed:<Space><LF>`
Results.load() will do a strip() that will remove the trailing space so parse_header() will fail to find `":<Space>"`

CasperSchippers, 2024-01-03:
Good catch!

This warrants the following consideration (relating also to what is mentioned in the PR about changing the separator): in principle all valid strings can be used as parameter name; while convenient, this does result in issues if you include the separator (i.e., `:<space>`) into the parameter name (or the value) (you can no longer load the value from file).
So whatever we decide regarding the separator, i'd be in favour of not allowing (and checking/enforcing) the parameter name to contain the separator.

BenediktBurger, 2024-01-03:
> So whatever we decide regarding the separator, i'd be in favour of not allowing (and checking/enforcing) the parameter name to contain the separator.

I agree.

poje42, 2024-01-03:
So what about this?
* Keep the current separator `:<space>` -> It will not break existing scripts
* Add checking of parameter names ->  Better robustness
* Remove strip in load() -> Parameters can have any content (almost, some characters like LF will not work)

BenediktBurger, 2024-01-03:
Sounds good to me.

CasperSchippers, 2024-01-03:
Sounds good indeed.

Seems like you found the strip that causes the issue (sorry didn't have time at the moment to try and find it). One thing that comes to mind is considering whether we should strip specifically for "\r\n" (instead of the standard strip which removes, from top of my head, "<space>\r\n") to prevent any windows vs Unix issues (not sure if such an issue would arise, but just wanted to mention it while I thought of it).

poje42, 2024-01-04:
The strip is done in load()
```
        with open(data_filename) as f:
            while not header_read:
                line = f.readline()
                if line.startswith(Results.COMMENT):
                    header += line.strip() + Results.LINE_BREAK
                    header_count += 1
                else:
                    header_read = True

```
Good comment @CasperSchippers , the strip must remain to remove other whitespace characters.
The new commit uses:
`                    header += line.strip('\t\v\n\r\f') + Results.LINE_BREAK
`

closed 2024-01-09 by BenediktBurger