pymeasure/pymeasure #1027: Rework Keithleys' next_error

BenediktBurger, 2024-01-15:
Actually, looking at the code, it seems to me that removes character `“` from string associated to error code.
It looks like something related to personal taste, on the other hand the ‘retry’ when reply length < 2, it seems important, if really needed.

It would be nice to remove completely this custom next_error to simplify code, but perhaps it is not possible. In any case, I am good to merge.

_Originally posted by @msmttchr in https://github.com/pymeasure/pymeasure/pull/1024#discussion_r1450822843_

msmttchr, 2024-01-15:
In my opinion, it’s difficult to find a reasonable motivation to justify the double quote removaI.
In addition, also custom `check_errors` could be probably removed.
Really, we need some volunteers with same instrument.

BenediktBurger, 2024-01-16:
> Really, we need some volunteers with same instrument.

For that reason I opened this issue.

msmttchr, 2024-01-22:
Found Keithkey 2420 in lab and did some testing with GPIB interface:
- Current style of error logging (customized): `INFO:pymeasure.instruments.keithley.keithley2400:Keithley 2400 reported error: -113, Undefined header`
- Standard style for error logging: `ERROR:pymeasure.instruments.instrument:Keithley 2400 SourceMeter: -113.0, "Undefined header"`

Generated in a loop 10000 errors and checked that empty error message or incorrect error are never detected.

I will try to test with serial interface and then I will submit a PR for removing the exception.

BenediktBurger, 2024-01-22:
Thanks for looking into it.

closed 2024-01-23 by msmttchr