pymeasure/pymeasure #1042: Deprecated telnetlib

BenediktBurger, 2024-02-08:
In #744 the telnetAdapter has been deprecated, as the telnetlib library is deprecated.

Having added codecov, it flags deprecated telnetlib in every PR.

I see two courses of action:
1. We mark that import somehow (maybe `pragma: no cover`?), such that it is not mentioned anymore.
2. We remove telnetAdapter from pymeasure altogether.

LongnoseRob, 2024-02-08:
A 3rd option might be to use pyvisa-py and pyserial with socket:// style URL-formatted resources.. Robert

BenediktBurger, 2024-02-08:
> A 3rd option might be to use pyvisa-py and pyserial with socket:// style URL-formatted resources

That is suggested by the deprecation notice.

I see, we should base the telnetlib adapter on the VISAAdapter.

The question is, whether it is worth the work.

bilderbuchi, 2024-02-15:
> Having added codecov, it flags deprecated telnetlib in every PR.

I don't think it's codecov that is flagging this import, because it also appears in the py3.11 run -- it's the Python interpreter I think (and it's correct in flagging it): https://github.com/pymeasure/pymeasure/actions/runs/7899943459/job/21560420178#step:7:205

Our deprecation of `TelnetAdapter` has been released for more than 6 months now (0.12.0). 
If the alternative would be significant maintenance work to avoid that warning on our CI, I'd be in favour of removing `telnetlib` with the next release. If we e.g. make instantiating `TelnetAdapter` throw an exception with clear instructions (i.e. the text of the current deprecation warning or the changelog entry), this will surface for any remaining users in a more actionable way than just an `ImportError`, which sounds good enough for me.

PR #1045 (2024-02-15): Remove TelnetAdapter
Closes #1042 

I chose a NotImplementedError, as the Adapter is not implemented (anymore).
closed 2024-02-15 by BenediktBurger