pymeasure/pymeasure #1049: Allow small drop in code coverage

BenediktBurger, 2024-02-19:
CI tests fail, if the code coverage is reduced, even only a little bit, maybe due to a deleted file.

We could allow a drop of half a percent (for example) without failing the CI status:

https://docs.codecov.com/docs/commit-status#target

CasperSchippers, 2024-02-21:
I did not realize before that a drop in coverage would result in a failed test.
Do we really want to _enforce_ the coverage or do we just want to _encourage_ keeping up the coverage?

I would be inclined to suggest to only include the coverage test as an encouragement, so no failing on reduced coverage, to keep contributing accessible.

(I have to admit that I also do very often not add tests for all code in my PRs since I do not think that testing every function or line of code is really necessary, and, especially for the GUI part, the testing is more difficult than the contribution itself.)

BenediktBurger, 2024-02-21:
> I did not realize before that a drop in coverage would result in a failed test.

There is an additional "check" marked as failed, if the requirements are not met.

> I would be inclined to suggest to only include the coverage test as an encouragement, so no failing on reduced coverage, to keep contributing accessible.

Until now, I ignored that failed test in several pull requests as we do not enforce it.

If we ignore that, we can disable that test altogether.

The pull request comment already encourages tests, so that should be sufficient.

BenediktBurger, 2024-02-22:
I changed the PR to do no status check, so coverage does not affect the CI pipeline.

PR #1052 (2024-02-27): Add codecov.yml with small coverage drop allowed.
This codecov.yml allows the coverage to drop by a quarter percent:
Sometimes the coverage fluctuates a little bit from test to test (around 0.1 percent), or removing lines reduces coverage, which makes that CI test fail. (see the last pull request #919 and current master)

closes #1049 

We could increase the threshold to half a percent or even one percent.
closed 2024-02-27 by CasperSchippers