pymeasure/pymeasure #1050: I think the PrologixAdapter isn't getting inherited correctly when initializing instruments that use channels in python 3.12.

Sionwage, 2024-02-21:
I am trying to create an instrument class for an Agilent E4418B Power Meter with the following code:

    import logging
    from time import sleep
    from time import time as now
    
    import numpy as np
    from pyvisa import VisaIOError
    
    from pymeasure.instruments import Instrument, Channel
    from pymeasure.instruments.validators import strict_discrete_set, truncated_range, strict_range
    
    log = logging.getLogger(__name__)
    log.addHandler(logging.NullHandler())
    
    class PowerMeterChannel(Channel):
        """Channel implimentation for the Agilent E4418B Power Meter"""
    
        frequency = Channel.control('SENS{ch}:FREQ?',
                                    'SENS{ch}:FREQ %e',
                                    """Control the frequency the power meter corrects its
                                    measurement for in Hz. Value range can be changed based
                                    on the power head used.
    
                                    Type: :code:`float`
    
                                    .. code-block:: python
    
                                        # set the frequency to 1.21GHz
                                        instr.frequency = 1.21e9
    
                                        if instr.frequency == 10e6:
                                            pass
    
                                    """,
                                    values=[10e6, 18e9],
                                    validator=strict_range,
                                    dynamic=True,
                                    cast=float)
    
    
    class AgilentE4418B(Instrument):
        """
        Represents the Agilent E4418B Power Meter for measuring RF/Microwave
        power at a given frequency.
    
        This device has one or two channels which would be displaced in one or two of the
        windows on the display.
    
        :param pymeasure.adapter adapter: Adapter used to connect to instrument
        :param str name: Name of the device or generated from the `AgilentE4418B.id` call on `__init__`
        :param includeSCPI: Keyword arguments for the adapter.
        :param kwargs: Keyword arguments for the adapter.
        """
    
        def __init__(
            self,
            adapter=None,
            name="Agilent E4418B Power Meter",
            **kwargs,
        ):
            super().__init__(adapter=adapter, name=name, includeSCPI=True, **kwargs)
    
    
    
        channels = Instrument.MultiChannelCreator(PowerMeterChannel, [1, 2])
    
        def ask(self, command, query_delay=None):
            self.adapter.write(command)
            if query_delay is not None:
                sleep(query_delay)
            return self.adapter.read()
    
        def read(self):
            return self.adapter.read()
    
        def write(self, command):
            self.adapter.write(command)
    
        id = Instrument.measurement(
            "*IDN?",
            """Get the identification of the instrument""",
            cast=str,
        )

If I don't define the `ask`, `read`, and `write` functions, I get this error when I call `id`:

`TypeError: 'NoneType' object is not callable`

Same for if I try to call one of the channels frequency.

    import pymeasure
    from pymeasure.adapters import PrologixAdapter
    from pymeasure.instruments.agilent.agilentE4418B import AgilentE4418B, PowerMeterChannel
    
    prologix = PrologixAdapter(resource_name="ASRL4::INSTR", visa_library="@py", auto=1)
    
    pm = AgilentE4418B(adapter=prologix.gpib(13))
    
    pm.id

returns `['HEWLETT-PACKARD', 'E4418A', '', 'A1.08.01']`

if I inspect `pm.ch_1.frequency` I get this:

    ---------------------------------------------------------------------------
    TypeError                                 Traceback (most recent call last)
    Cell In[3], line 1
    ----> 1 pm.ch_1.frequency
    
    File ~\OneDrive\Documents\Python Git\pymeasure\pymeasure\instruments\common_base.py:298, in CommonBase.__getattribute__(self, name)
        295     if name in self._special_names:
        296         raise AttributeError(
        297             f"{name} is a reserved variable name and it cannot be read")
    --> 298 return super().__getattribute__(name)
    
    File ~\OneDrive\Documents\Python Git\pymeasure\pymeasure\instruments\common_base.py:79, in DynamicProperty.__get__(self, obj, objtype)
         77     if hasattr(obj, attr_instance_name):
         78         kwargs[attr] = getattr(obj, attr_instance_name)
    ---> 79 return self.fget(obj, **kwargs)
    
    File ~\OneDrive\Documents\Python Git\pymeasure\pymeasure\instruments\common_base.py:555, in CommonBase.control.<locals>.fget(self, get_command, values, map_values, get_process, command_process, check_get_errors)
        553 if get_command is None:
        554     raise LookupError("Property can not be read.")
    --> 555 vals = self.values(command_process(get_command),
        556                    separator=separator,
        557                    cast=cast,
        558                    preprocess_reply=preprocess_reply,
        559                    maxsplit=maxsplit,
        560                    **values_kwargs)
        561 if check_get_errors:
        562     try:
    
    File ~\OneDrive\Documents\Python Git\pymeasure\pymeasure\instruments\common_base.py:406, in CommonBase.values(self, command, separator, cast, preprocess_reply, maxsplit, **kwargs)
        389 def values(self, command, separator=',', cast=float, preprocess_reply=None, maxsplit=-1,
        390            **kwargs):
        391     """Write a command to the instrument and return a list of formatted
        392     values from the result.
        393 
       (...)
        404     :returns: A list of the desired type, or strings where the casting fails.
        405     """
    --> 406     results = self.ask(command, **kwargs).strip()
        407     if callable(preprocess_reply):
        408         results = preprocess_reply(results)
    
    File ~\OneDrive\Documents\Python Git\pymeasure\pymeasure\instruments\common_base.py:387, in CommonBase.ask(self, command, query_delay)
        385 self.write(command)
        386 self.wait_for(query_delay)
    --> 387 return self.read()
    
    File ~\OneDrive\Documents\Python Git\pymeasure\pymeasure\instruments\channel.py:79, in Channel.read(self, **kwargs)
         77 def read(self, **kwargs):
         78     """Read up to (excluding) `read_termination` or the whole read buffer."""
    ---> 79     return self.parent.read(**kwargs)
    
    TypeError: 'NoneType' object is not callable

I still have to go back to python 3.11 and see if its related to 3.12. Commenting out the read, write, ask definitions causes the same typeerror for just calling `id`.

BenediktBurger, 2024-02-21:
I don't see the problem yet. In #1051 I started to update the CI in order to check for problems automatically.

BenediktBurger, 2024-02-21:
Could you try your new instrument with a Protocol Test, please? That should eliminate communication issues with the device itself and limit the problems to pymeasure itself.

BenediktBurger, 2024-02-21:
In #1051 tests for Python 3.12 pass (except for the nicaiu problem you mentioned in the other issue)

Sionwage, 2024-02-21:
Holy smokes, I'm the idiot. 

I found my issue which was completely unrelated to pymeasure. I was blindly writing properties in into the class and one of them was `read` for `READ[1|2]` but didn't catch because of flipping back and forth in my git branches. 

[image]

[image]

Everything works and I feel stupid for publishing this for the internet.

closed 2024-02-21 by Sionwage

BenediktBurger, 2024-02-21:
I'm glad you found the issue.

It happens to the best.