pymeasure/pymeasure #1061: FutureWarning by pandas when using TableWidget

pyZerrenner, 2024-03-11:
The lines 176 and 177 in the table_widget.py script generate a `FutureWarning` by Pandas. This is particularly annoying, since the lines are executed multiple times a second when using a `TableWidget` and quickly fill up the console.

```
C:\<...>\miniforge3\envs\test\Lib\site-packages\pymeasure\display\widgets\table_widget.py:176: FutureWarning: Series.__getitem__ treating keys as positions is deprecated. In a future version, integer keys will always be treated as labels (consistent with DataFrame behavior). To access a value by position, use `ser.iloc[pos]`
  value = results.data.iloc[row][col]
C:\<...>\miniforge3\envs\test\Lib\site-packages\pymeasure\display\widgets\table_widget.py:177: FutureWarning: Series.__getitem__ treating keys as positions is deprecated. In a future version, integer keys will always be treated as labels (consistent with DataFrame behavior). To access a value by position, use `ser.iloc[pos]`
  column_type = results.data.dtypes[col]
```

The warnings can be avoided by replacing the lines with (though there is probably a more concise solution for the first line):
```python
value = results.data.iloc[row].iloc[col]
column_type = results.data.dtypes.iloc[col]
```

My current workaround (aside from editing table_widget.py myself) is to suppress `FutureWarning` by including
```python
import warnings
warnings.simplefilter(action='ignore', category=FutureWarning)
```

<details>

<summary>Here is a MWE to generate the warning</summary>

```python
r"""
MWE for FutureWarning by pandas when using TableWidget in PyMeasure 0.13.1

Conda setup: ``conda create -n test -c conda-forge pymeasure pyqt``

Hit the 'Queue' button to start the procedure and see this warning in the
console:

C:\<...>\Lib\site-packages\pymeasure\display\widgets\table_widget.py:176:
    FutureWarning: Series.__getitem__ treating keys as positions is deprecated.
    In a future version, integer keys will always be treated as labels
    (consistent with DataFrame behavior). To access a value by position,
    use `ser.iloc[pos]`
  value = results.data.iloc[row][col]
C:\<...>\Lib\site-packages\pymeasure\display\widgets\table_widget.py:177:
    FutureWarning: Series.__getitem__ treating keys as positions is deprecated.
    In a future version, integer keys will always be treated as labels
    (consistent with DataFrame behavior). To access a value by position,
    use `ser.iloc[pos]`
  column_type = results.data.dtypes[col]
"""

from time import sleep
from random import random

import sys
import tempfile

from pymeasure.display.Qt import QtWidgets
from pymeasure.experiment import Procedure
from pymeasure.experiment import Results
from pymeasure.display.widgets import TableWidget
from pymeasure.display.windows.managed_window import ManagedWindow

class SimpleProcedure(Procedure):
    DATA_COLUMNS = ['iter', 'rand']

    def execute(self):
        # run for ~5 seconds & send random values
        for i in range(10):
            self.emit('results', {'iter': i,
                                  'rand': random()})
            sleep(0.5)

[... 88 lines cut ...]
python-dateutil           2.9.0              pyhd8ed1ab_0    conda-forge
python-tzdata             2024.1             pyhd8ed1ab_0    conda-forge
python_abi                3.12                    4_cp312    conda-forge
pytz                      2024.1             pyhd8ed1ab_0    conda-forge
pyvisa                    1.14.1             pyhd8ed1ab_0    conda-forge
qt-main                   5.15.8              h9e85ed6_19    conda-forge
setuptools                69.1.1             pyhd8ed1ab_0    conda-forge
sip                       6.7.12          py312h53d5487_0    conda-forge
six                       1.16.0             pyh6c4a22f_0    conda-forge
tbb                       2021.11.0            h91493d7_1    conda-forge
tk                        8.6.13               h5226925_1    conda-forge
toml                      0.10.2             pyhd8ed1ab_0    conda-forge
tomli                     2.0.1              pyhd8ed1ab_0    conda-forge
typing_extensions         4.10.0             pyha770c72_0    conda-forge
tzdata                    2024a                h0c530f3_0    conda-forge
ucrt                      10.0.22621.0         h57928b3_0    conda-forge
vc                        14.3                hcf57466_18    conda-forge
vc14_runtime              14.38.33130         h82b7239_18    conda-forge
vs2015_runtime            14.38.33130         hcb4865c_18    conda-forge
wheel                     0.42.0             pyhd8ed1ab_0    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
zstd                      1.5.5                h12be248_0    conda-forge
```

</details>

BenediktBurger, 2024-03-11:
Thanks for raising this issue. @CasperSchippers , @msmttchr could you take a look, whether the first suggestion (with the `iloc`) would fix that problem without causing issues for older versions?

CasperSchippers, 2024-03-11:
I've had a look and think that the suggestion with (`iloc`) should indeed work with older versions. I think that in principle that way should've been preferred all along when indexing by integer position. Technically, since we're indexing for a single scalar value, I thin that technically the recomended indexing method would be `iat` (which is alledgeldy a lot faster, not sure if we'd notice though).

CasperSchippers, 2024-03-11:
Opened a short PR that fixes this warning (using `iat`, if someone has a preference for `iloc`, that'd also be fine with me)

BenediktBurger, 2024-03-11:
Thanks for fixing.

closed 2024-03-11 by CasperSchippers

pyZerrenner, 2024-03-11:
Thanks for the quick response 👍

PR #1062 (2024-03-11): Fix/pandas futurewarning
Fixes #1061

Proposes to use `iat` to replace `__getitem__` indexing for indexing by position. Since we're indexing for a single scalar value, I thinkt the recommended method is `iat`, rather than `iloc` (https://pandas.pydata.org/docs/user_guide/indexing.html#fast-scalar-value-getting-and-setting)