pymeasure/pymeasure #1063: Multiple procedures with plotter

MiguelBL248, 2024-03-12:
Hello everybody,

I have a script that communicates with an AgilentXGS600. In this driver I had a function named read_all_pressures(), that returns an array of two pressures measured with the device ([p1, p2]). This driver is available on the internet.

What I want is to live plot these two values of the pressure as a function of the time (or just iterations, I just want to see the shape of the curves). I've been trying with Plotter but I do not know how exactly to implement two graphs instead of just one. There is an issue from 2020 that talks about a 'MultiPlotter' class, but I haven't quite implemented it properly. 

Thanks in advance.

BenediktBurger, 2024-03-13:
For that you do not need several procedures.
You can just store the two values as two variables in the same procedure.

Maybe the tutorial for the https://pymeasure.readthedocs.io/en/latest/tutorial/graphical.html#using-the-manageddockwindow helps

MiguelBL248, 2024-03-13:
I've been checking the manageddock tutorial, but when I run the code the window appears, but I am not able to plot the curves of my two procedures. Maybe I'm lost at an obvious point but I can't figure out how to make it run properly.

BenediktBurger, 2024-03-13:
Could you share your code in here?

You do not need two procedures, but just a single one, something along those lines:
~~~python
from time import sleep, perf_counter
from pymeasure.experiment import Procedure
from pymeasure.experiment import IntegerParameter

class SimpleProcedure(Procedure):

    # a Parameter that defines the number of loop iterations
    iterations = IntegerParameter('Loop Iterations')

    # a list defining the order and appearance of columns in our data file
    # COMMENT: here you define the columns
    DATA_COLUMNS = ['Iteration', 'time']

    def execute(self):
        """Execute the procedure.

        Loops over each iteration and emits the current iteration,
        before waiting for 0.01 sec, and then checking if the procedure
        should stop.
        """
        for i in range(self.iterations):
            # COMMENT: in this data, you add the values for the different columns, here for the iteration and for the time:
            data = {'Iteration': i, 'time': perf_counter()}
            self.emit('results', data)
            sleep(0.01)
            if self.should_stop():
                break
~~~

I made two changes (and comments), how to store more than one variable in a single procedure, maybe that helps as a starting point.

MiguelBL248, 2024-03-20:
Thank you very much for your answer. Finally, I managed to use the the ManagedDockWindow class to my purpose.

BenediktBurger, 2024-03-20:
I'm glad, that it works for you.
If your question is answered, please close this issue.