pymeasure/pymeasure #1076: Running many scans causes the system to crash.

OrionSmedley, 2024-03-23:
If we run 500 scans, pymeasure crashes. 
In the device manager, it says that the python memory usage is several gigabits.

This seems to be consistent with what other experiments see, elsewhere in my lab, which makes me suspect that it's an issue with pymeasure, rather than the specific code for the individual experiments.

Each scan file only takes up 11kb, so it's not that the quantity of data would be an issue, even if all of the scans were in the ram at the same time. Maybe there's some cache for the graphs in the QT gui interface window?

BenediktBurger, 2024-03-23:
I suspect also, that it is too much plotting all the graphs.

In my measurement software, Pyqtgraph (the plotting library, also used by pymeasure) can't keep up plotting the new data, once I have too many data points, especially, if I plot them as dots (graphically small circles) instead of a line.

OrionSmedley, 2024-03-23:
I agree, though it seems like even if you clear all the graphs, the information is still cached somewhere. Because (after all the scans finished) the memory usage did not change after hitting clear all 
[image]

msmttchr, 2024-03-23:
Is it possible that the memory consumption doesn't come from pymeasure? I mean something added in the test procedure?

OrionSmedley, 2024-03-23:
It's possible, just it would mean that separate programs used in different parts of the lab made the same mistake.
Is there an easy way to check that?

msmttchr, 2024-03-23:
> It's possible, just it would mean that separate programs used in different parts of the lab made the same mistake.
> 
> Is there an easy way to check that?

I would try replacing the experiment specific part with some emulation which generates similar amount of data and then check that you have similar problems. Or use the tutorial example possibly with some modifications to replicate the problem.