pymeasure/pymeasure #1078: The Browser's "Delete Data File" action only removes the first data file

pyZerrenner, 2024-03-27:
When supplying multiple file names to a `Results` instance and running the experiment with a `ManagedWindow`, all files are created and filled with the same data. When right-clicking the corresponding browser entry after the experiment finished and selecting the "Delete Data File" action, only the very first data file gets deleted. The other data files remain. Is this an intended behaviour (since the action not in plural and only one file name is shown)?

The reason is, that `ManagedWindowBase.delete_experiment_data` unlinks `experiment.data_filename` and does not loop over `experiment.data_filenames`.

<details>
<summary> Code of `ManagedWindowBase.delete_experiment_data`</summary>

https://github.com/pymeasure/pymeasure/blob/1c8d1b215481b88bbb902e43caeb97037ed423e9/pymeasure/display/windows/managed_window.py#L365-L373

</details>

<details>
<summary> MWE for testing </summary>

```python
"""
MWE for deleting an experiment with mulitple data files from the browser
in PyMeasure 0.13.1

Start the program and run an experiment. Two data files are created in the
current directory. After the experiment finished, right-click its entry in the
browser and select "Delete Data Files". Only the first data file is deleted.
"""
from time import sleep
from random import random

import sys, os

from pymeasure.display.Qt import QtWidgets
from pymeasure.experiment import Procedure
from pymeasure.experiment import Results
from pymeasure.display.windows.managed_window import ManagedWindow

class SimpleProcedure(Procedure):
    DATA_COLUMNS = ['iter', 'rand']

    def execute(self):
        # run for ~5 seconds & send random values
        for i in range(10):
            self.emit('results', {'iter': i,
                                  'rand': random()})
            sleep(0.5)

class MainWindow(ManagedWindow):
    def __init__(self):
        super().__init__(
            procedure_class=SimpleProcedure,
            x_axis='iter',
            y_axis='rand'
            )
        self.setWindowTitle('MWE Delete Multiple Data Files From Browser')

    def queue(self):
        # save two data files into the current directory
        filename = [os.path.abspath('datafile_01.csv'),
                    os.path.abspath('datafile_02.csv')]

        procedure = self.make_procedure()
        results = Results(procedure, filename)
        experiment = self.new_experiment(results)

        self.manager.queue(experiment)

if __name__ == "__main__":
    app = QtWidgets.QApplication([])
    window = MainWindow()
    window.show()
    sys.exit(app.exec())
```

</details>

CasperSchippers, 2024-05-06:
Sorry for the late response.

I don't think this is intended behaviour, as I don't think (but maybe I'm wrong) there was an intended behaviour for multiple files when the 'remove datafile' option was added. I'm at this moment also not sure what the expected/most reasonable behaviour is; I would think that removing all files would make more sense, but can imagine that other people might think othwerise, maybe depending on the reason for using multiple files.

Could you tell me, what is the reason to write two identical datafiles at the same time? (your reason, or in general)

pyZerrenner, 2024-05-07:
I actually do not use multiple files as defined by pymeasure and I am also not quite sure, what a use case for writing multiple identical files would be. Maybe for a backup?

I stumbled over this inconsistency when implementing temporary files in my own script: Measurement files that are identified as "temporary" get deleted automatically when the program is closed but also when a measurement is removed from the browser. That was when I came across this issue in `ManagedWindowBase.delete_experiment_data`.

CasperSchippers, 2024-05-07:
Ahh, that makes sense; thanks for making this issue known.

Whatever we choose to do, probably most important is to document the behaviour.

We can either choose to keeep it as is, and delete only the first file, or opt to delete all files. Maybe an ideal situation would be to be able to select ad hoc which file(s) to delete, if mulitple files are attached; but this would probably be more work than how much it would benefit; a nice middle road may be to have to delete options, one for first and one for all. A last alternative would be to fully remove the multiple-file feature if we don't know what it is used for, but I don't think this would be a good idea.

Either case I think most import would be to clarify/document/tooltip the behaviour.