pymeasure/pymeasure #1093: Graphical Interface Tutorials not working, ".queue" not implemented

bauzbue, 2024-04-16:
Trying out the tutorials, as soon as I press the "queue" button in the GUI (f.e. on the first tutorial of "ManagedWindow"), i get the error: 

  File "c:\...\Programs\Python\Python312\Lib\site-packages\pymeasure\display\windows\managed_window.py", line 523, in queue
    raise NotImplementedError(NotImplementedError: Abstract method ManagedWindow.queue not implemented)

A possible workaround is to append: 
    _def queue(self):
    filename =  "testest" # unique_filename('results', prefix="data") # from pymeasure.experiment

    procedure = self.make_procedure() # Procedure class was passed at construction
    results = Results(procedure, filename)
    experiment = self.new_experiment(results)

    self.manager.queue(experiment)_
   
at the end of the class ManagedWindow in the file "managed_window.py"

I am just starting with pyMeasure

BenediktBurger, 2024-04-17:
Which pymeasure version do you use? A released version (installed from pypi via pip or via conda) or the latest github version?

@CasperSchippers could that be related to one of your last changes?

bauzbue, 2024-04-17:
Hi Benedikt, 
I use currently the version PyMeasure 0.13.1, installed via pip on python 3.12.2

Before I tried with an anaconda installed version, however I had issues to install qtpy (or PySide6).

CasperSchippers, 2024-04-17:
This is definitely related to recent changes: I implemented a default `queue` method such that users (unless really required) don't need to (PR #964). This is however after the latest release (0.13.1); issue is that the ['latest' documentation version](https://pymeasure.readthedocs.io/en/latest/) reflects the pymeasure version on GitHub, not the released version. For the released version, you should refer to the ['stable' version of the documentation](https://pymeasure.readthedocs.io/en/stable/) (where the [graphical](https://pymeasure.readthedocs.io/en/stable/tutorial/graphical.html) section does still show implementing a queue method).

If you want to use the newest (GitHub) version, you should (especially when using pip, and given that you do have git installed on your system) be able to get the GitHub version using
```commandline
pip install git+https://github.com/pymeasure/pymeasure.git
```

This will have the advantage of not having to write a queue method yourself and has some nice input-fields for the file and file-location. If you use the latest release version, you do have to implement these yourselves.

Two closing notes:
- I can imagine (at least I think it would be a good idea, seeing the numpy2 thingy) having a new release somewhere soon.
- We might have to think about how this documentation is presented (stable vs latest), since it might be a bit confusing.

bauzbue, 2024-04-17:
Hi Casper, 

Thanks a lot for the clarification. I will for now focus on the stable version of the docs, which solved my issue. 

For me this topic is closed.

closed 2024-04-17 by bauzbue

CasperSchippers, 2024-06-26:
Might be interesting to know that with the latest release (0.14), the above item in the documentation is now correct.

There is a default queue method now (and a file name and location input field), so for most use cases no need to implement one yourself.