pymeasure/pymeasure #1096: CI fails on macos

BenediktBurger, 2024-04-29:
In several PRs and the main branch, the CI fails on all macos runners, but succeeds on the other OS and all python versions:
- #1070 
- #1092 

Error message in the `Install pymeasure requirements` step:
~~~
Run mamba-org/setup-micromamba@v1
Will download micromamba to /Users/runner/micromamba-bin/micromamba
Install micromamba
Adding conda-forge to condarc channels ...
Initialize micromamba for bash.
Restoring environment `pymeasure` from `/Users/runner/micromamba/envs/pymeasure` ...
Install environment `pymeasure`
  /Users/runner/micromamba-bin/micromamba create -y -r /Users/runner/micromamba -f .github/pymeasure.yml python=3.10 --log-level warning --rc-file /Users/runner/work/_temp/setup-micromamba/.condarc
  error    libmamba Could not solve for environment specs
      The following package could not be installed
      └─ pyvisa 1.12.0**  does not exist (perhaps a typo or a missing channel).
  critical libmamba Could not solve for environment specs
  Error: The process '/Users/runner/micromamba-bin/micromamba' failed with exit code 1
~~~

It seems to be related to pyvisa.

@bilderbuchi it happens also in another CI

mcdo0486, 2024-04-29:
Added a PR that should fix this. In short, M1 apple silicon were added at some point to the github runners pool. The previous pyvisa version 1.12.0 was only for x64 macos. Version 1.13.0 uses the noarch version which works on M1 CPU.

https://github.com/pymeasure/pymeasure/pull/1097

CasperSchippers, 2024-04-29:
Indeed @mcdo0486, that has fixed it. Thanks a lot!