pymeasure/pymeasure #1106: CI fails due to timing issue occasionally

BenediktBurger, 2024-05-10:
Quite often one test fails due to timing issues making the whole CI pipeline fail, probably the time limit should be reduced slightly:
~~~
    def test_check_query_delay():
        with expected_protocol(ATSBase, [("TTIM?", "7")]) as inst:
            start = perf_counter()
            assert inst.maximum_test_time == 7
            delay = perf_counter() - start
>           assert delay > 0.05
E           assert 0.0491846000000038 > 0.05

tests\instruments\temptronic\test_temptronic_base.py:36: AssertionError
~~~

CasperSchippers, 2024-05-13:
It _sounds_ a bit like a floating-point errors: the `start` and `delay` might be too large (numericall) such that that difference cannot be calculated accurately.

BenediktBurger, 2024-05-14:
Another solution could be `pytest.approx`, however, we should check first, what is actually tested, maybe the test is ok, but the code under test not.

BenediktBurger, 2024-05-14:
Having merged #1109, CI runs succeed. After the previous two merges, they always failed.

Seems to work, we should keep it open though for some more time.

BenediktBurger, 2024-05-15:
It did not work in [CI run](https://github.com/pymeasure/pymeasure/actions/runs/9091306319/job/24985650504?pr=1039):
~~~
    def test_check_query_delay():
        with expected_protocol(ATSBase, [("TTIM?", "7")]) as inst:
            start = perf_counter_ns()
            assert inst.maximum_test_time == 7
            delay = perf_counter_ns() - start
>           assert delay > 0.05 * 1e9
E           assert 47828800 > (0.05 * 1000000000.0)
~~~

PR #1110 (2024-05-21): Fix temptronic test timing issue
Closes #1106 

This should solve the problem: reducing the time limit.
It's brute force, but I do not see any other option.
closed 2024-05-21 by BenediktBurger