pymeasure/pymeasure #1111: Define fallback version in setup.py

BenediktBurger, 2024-05-16:
Hello,
I actually run into another issue when trying to install from the cloned repository.

If you do so: 
```
pip install -e .
```

and if your shell is not aware of git (which was my case) then the installation complains it cannot find the version because it cannot find git. It troubled me for quite some time before I realized this. So if you install using pip the repo code (from a zip) of from a clone but with unaware git app, the installation will fail...

_Originally posted by @seb5g in https://github.com/pymeasure/pymeasure/discussions/893#discussioncomment-9459215_

Should we define a fallback version for setuptools_scm?
According to https://github.com/pypa/setuptools_scm/issues/549, the fallback is for authors (us), while the environment variable is for packages (or users).

seb5g, 2024-05-16:
Well I'm not sure why you use scm to define your version and not hardcode it into a file. Is there a reason?
Then where is the version specified within git?

BenediktBurger, 2024-05-17:
> Well I'm not sure why you use scm to define your version and not hardcode it into a file. Is there a reason? Then where is the version specified within git?

Setuptools_scm reads the "source code management" (here git) to deduce a version info. In git it reads the last tag in the form "vMajor.Minor.Patch...".

Why someone wants to use it: you do not have to change a file and add a tag.
The tag (created on release) is sufficient.

The reason why pymeasure did it, I do not know, as it predates my participation. I guess it is the above mentioned reason.

seb5g, 2024-05-17:
well it makes sense indeed, I think my particular case where i use git wrapped into github desktop (and not shell aware) is a bit particular to make a modification of the code for this. However a fallback version like 0.0.0 would be nice (it is actually what happens when I removed the setuptools_scm bits within the pyproject.toml for the installation to work)

bilderbuchi, 2024-05-22:
The original reason was a user request for a more accurate/precise version (e.g. when living on pymeasure master branch and working in notebooks, if I remember correctly). That we do not need to manually update some version file is an additional benefit.

If we get a precise description of what is going wrong in the OP's case, maybe we can adapt the logic in https://github.com/pymeasure/pymeasure/blob/master/pymeasure/__init__.py#L30 to cover that case and fall into the fallback branch.