pymeasure/pymeasure #1112: Setuptools_scm version fails due to invalid time format

BenediktBurger, 2024-05-21:
Upstream issue: https://github.com/pypa/setuptools_scm/issues/1038

CI runs fail in the "pymeasure version" step under Python 3.9 with
~~~
Run python -c "import pymeasure;print(pymeasure.__version__)"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/runner/work/pymeasure/pymeasure/pymeasure/__init__.py", line 35, in <module>
    __version__ = setuptools_scm.get_version(root='..', relative_to=__file__)
  File "/home/runner/micromamba/envs/pymeasure/lib/python3.8/site-packages/setuptools_scm/_get_version_impl.py", line 155, in get_version
    maybe_version = _get_version(config, force_write_version_files=True)
  File "/home/runner/micromamba/envs/pymeasure/lib/python3.8/site-packages/setuptools_scm/_get_version_impl.py", line 93, in _get_version
    parsed_version = parse_version(config)
  File "/home/runner/micromamba/envs/pymeasure/lib/python3.8/site-packages/setuptools_scm/_get_version_impl.py", line 56, in parse_version
    or parse_scm_version(config)
  File "/home/runner/micromamba/envs/pymeasure/lib/python3.8/site-packages/setuptools_scm/_get_version_impl.py", line 35, in parse_scm_version
    return _entrypoints.version_from_entrypoint(
  File "/home/runner/micromamba/envs/pymeasure/lib/python3.8/site-packages/setuptools_scm/_entrypoints.py", line 55, in version_from_entrypoint
    maybe_version: version.ScmVersion | None = fn(root, config=config)
  File "/home/runner/micromamba/envs/pymeasure/lib/python3.8/site-packages/setuptools_scm/git.py", line 211, in parse
    return _git_parse_inner(
  File "/home/runner/micromamba/envs/pymeasure/lib/python3.8/site-packages/setuptools_scm/git.py", line 272, in _git_parse_inner
    node_date = wd.get_head_date() or datetime.now(timezone.utc).date()
  File "/home/runner/micromamba/envs/pymeasure/lib/python3.8/site-packages/setuptools_scm/git.py", line 131, in get_head_date
    return res.parse_success(
  File "/home/runner/micromamba/envs/pymeasure/lib/python3.8/site-packages/setuptools_scm/_run_cmd.py", line 77, in parse_success
    return parse(self.stdout)
  File "/home/runner/micromamba/envs/pymeasure/lib/python3.8/site-packages/setuptools_scm/git.py", line 121, in parse_timestamp
    return datetime.fromisoformat(timestamp_text).date()
ValueError: Invalid isoformat string: '2024-05-21T07:40:32Z'
~~~

Solution could be bumping setuptools_scm version. However, in https://github.com/pymeasure/pyleco/pull/81 I tried this, but it did not work as expected: Even runners with a more recent python version failed.
Another (maybe temporary) solution is to remove printing the pymeasure version.

bilderbuchi, 2024-05-21:
> However, in [pymeasure/pyleco#81](https://github.com/pymeasure/pyleco/pull/81) I tried this, but it did not work as expected: Even runners with a more recent python version failed.

I don't see what happened, probably because of a rebase, but was that with setuptools 8.1.0 that is supposed to contain the fix?

bilderbuchi, 2024-05-21:
If we remove the printing of the version, doesn't that still fail for users doing just that?

bilderbuchi, 2024-05-21:
> I don't see what happened, probably because of a rebase, but was that with setuptools 8.1.0 that is supposed to contain the fix?

If the fix to https://github.com/pypa/setuptools_scm/issues/1038 is indeed incomplete, I think we should report that upstream?
First we have to make sure that that's really what's happening, though (e.g. ensure that we really get that new `setuptools_scm` version on those runners) . 
Should we bump the version in #1104 and try to diagnose?

BenediktBurger, 2024-05-21:
> I don't see what happened, probably because of a rebase, but was that with setuptools 8.1.0 that is supposed to contain the fix?

Yes, I rebased it, as it did not work and was not related to the content of the PR anyways.

> First we have to make sure that that's really what's happening, though (e.g. ensure that we really get that new setuptools_scm version on those runners) .

what do you suggest to make that sure? I added it to the pyproject.toml setup requirements.

bilderbuchi, 2024-05-21:
e.g. check the mamba list output from CI to confirm the version.

BenediktBurger, 2024-05-21:
In #1104 I noticed, that it is necessary to enforce the setuptools_scm version in the environment file, it is not sufficient to write it in the pyproject.toml (and even deleting all the cached environments!).

Probably we should reverse the enforcement later on, once the caches contain the version 8.1.0 (or newer).

bilderbuchi, 2024-05-22:
Fair enough. 
Maybe we actually need to add `setuptools_scm` to the runtime dependencies (`install_requires`) in `setup.cfg`, not only the `build-system`, as this part of the dependency is also used at runtime.

BenediktBurger, 2024-05-22:
> Maybe we actually should add `setuptools_scm` to the runtime dependencies (`install_requires`) in `setup.cfg`, not only the `build-system`, as this part of the dependency is also used at runtime.

We catch it with an try clause. It is only used, if it is installed.
This reduces the dependency for installs via pypi.

bilderbuchi, 2024-05-22:
Ah, yes, thanks for reminding me. Then this is fine as we have it now, I think.