pymeasure/pymeasure #1113: List of dll drivers as new webpage

BenediktBurger, 2024-06-03:
An additional idea:
We could maintain a directory of dll drivers similar to your overview of measurement software, @bilderbuchi.
I already added a few dll driver repos to that overview, but they don't fit that well.

_Originally posted by @BenediktBurger in https://github.com/pymeasure/pymeasure/issues/1058#issuecomment-2143493061_
            

Maybe we can create a pymeasure badge and invite the projects hosted in this list to fly the pymeasure badge, thus linking the drivers to pymeasure.

seb5g, 2024-06-03:
I actually wrote a few drivers around some dlls and would like to have them on pymeasure as at the moment thay are within pymodaq plugins and are not really visible.

I recently used this package: https://msl-loadlib.readthedocs.io/en/stable/ which is a wrapper around dlls (C, C++ dotnet...) that also allows to easily use 32bits dlls on a 64bit python instance. I'm wondering if this could be a way to add some structure in the dll drivers?

BenediktBurger, 2024-06-03:
Thanks for that library. A colleague needs exactly that bitness translation for a current project (old camera).

> I'm wondering if this could be a way to add some structure in the dll drivers?

Which structure do you mean?

seb5g, 2024-06-03:
well like a help to introduce some coding standard as you already do with pymeasure. The dll loading could be hidden behind the use of such a library and the call of the functions would also be easier.