pymeasure/pymeasure #1120: How to give multiple inputs with instrument.control method?

Ashwolaa, 2024-06-11:
Hi,

I am currently setting up an experiment and will use the LakeShore340.
I have found that a library was already developed in pymeasure but I would like to extend it.

To be more precise, I would like to add to the LakeShoreHeaterChannel class a new method which defines the loop parameter. I add the documentation in the following picture. 
[image]

My problem comes from the Instrument.control method. I mostly saw examples in which a single input was given. 
I achieve to give multiple inputs by using the following syntax
`lakeshore.output_1.parameters = ('B',1,2,1)`
with the following method
`parameters = Instrument.control(
        'CSET? {ch}',
        'CSET {ch},%s,%i,%i,%i',
        """Setup controle loop parameters""",
        )            `
However, I am struggling to write the validator part since there is now multiple inputs.

Thanks!

BenediktBurger, 2024-06-11:
Thanks for asking.

First, you do not have to use `control`. In some cases a method is better to understand.

> lakeshore.output_1.parameters = ('B',1,2,1)

Yes, that is the syntax to give a tuple of several values to the property.

> However, I am struggling to write the validator part since there is now multiple inputs.

You do not have to use a validator.
If you want to use one, you have to write your own validator, which is in the form `SpecialValidator(value, values)`, where `value` is the tuple you give to the property and `values` is the values parameter of `control`. I suggest, that you ignore the `values` parameter and make all the validation logic in that validator itself.

Ashwolaa, 2024-06-11:
Thanks for the answer, the validator was mostly to follow what was already implemented in the class. 

I will see if it is worth going for it then.

closed 2024-06-14 by Ashwolaa