pymeasure/pymeasure #1130: id( ) command fails for Agilent B1500 SDA

AdMico, 2024-06-25:
Command gives:

TypeError: 'str' object is not callable

Might want to name it something other than id?

msmttchr, 2024-06-25:
`id` is a property which returns the output of scpi command `*IDN?`, so you should not add round brackets as you would when calling a method.

BenediktBurger, 2024-06-26:
You just use `device_ID = inst.id` to assign the ID to some variable. You use all these properties like variables of the device, not like methods.

AdMico, 2024-06-27:
OK, then I'd say it's a documentation error, as the brackets I had just came directly from the documentation.

msmttchr, 2024-06-27:
> OK, then I'd say it's a documentation error, as the brackets I had just came directly from the documentation.

Could you please send some reference where the documentation error is ?

AdMico, 2024-06-27:
https://pymeasure.readthedocs.io/en/latest/api/instruments/agilent/agilentB1500.html

About halfway down the Command Translation list in General Information.

msmttchr, 2024-06-27:
Indeed, you are right. It looks like a documentation bug. If you feel comfortable, you can open a PR for it, otherwise I will try to do it on my side. Thanks for spotting this problem.

AdMico, 2024-06-27:
Might be better that I let you do that one (unless you can point me at a good resource for how to do it) as I haven't done that before and don't know where to start.

closed 2024-07-01 by msmttchr