pymeasure/pymeasure #1131: Questions about running the NI-DAQ packages in pymeasure

EliasDubroff, 2024-06-27:
Hello,

I've been trying to use the NI packages in PyMeasure with my NI USB-6210 DAQ, but I have not been able to get them to work, no matter how much I try. While searching through the GitHub issues section for these NI-DAQ packages, I came across a few posts discussing how the NI instruments no longer work on Python 3.12 due to issues with the C API, 'nicaiu', particularly within the daqmx.py file. I am not an extremely experienced programmer so I am not very knowledgeable about issues such as these, and thought it would be best to make my own post in hopes of clarifying what was going on with these packages.

I would like to ask if these scripts are still currently unusable with PyMeasure on Python 3.12. If so, are there other alternative methods that I could use to connect to my device within PyMeasure? Would it also be possible to go back to a previous version of python in order to get them to work correctly?

Some advice with this issue would be greatly appreciated. Thank you very much!

BenediktBurger, 2024-06-27:
Hello, thanks for asking.

Did you install the nicaiu package?
I personally use the nidaqmx package without going through pymeasure.

> Would it also be possible to go back to a previous version of python in order to get them to work correctly?

Yes, you can install another [python version](https://www.python.org/downloads/).

EliasDubroff, 2024-06-27:
Hi there,

I'm unsure as to what you mean by the nicaiu package as it is a .dll? I have the nidaqmx package installed on my computer, I cannot find a separate nicaiu package after doing a quick search. If there is one could you please point me as to where/how I can install that?

BenediktBurger, 2024-06-27:
> Hi there,
> 
> I'm unsure as to what you mean by the nicaiu package as it is a .dll? I have the nidaqmx package installed on my computer, I cannot find a separate nicaiu package after doing a quick search. If there is one could you please point me as to where/how I can install that?

Sorry, I did not remember it correctly. You're right, it is a dll.

Regarding the current state:
We know, that python3.12 makes problems with that nicaiu if it is not installed.
We do not know, whether it works with python 3.12.
As it requires a dll, we cannot test it without having the dll and a corresponding device.

Pymeasure is a great package, but dll driven drivers are not its focus (the ni part is an odd historical relict).

EliasDubroff, 2024-06-27:
Got it, that all makes sense. To me it seems like the best course of action for me now would be for me to create my own class in Pymeasure in order to be able to connect with my device, as I would like to automate this device along with others which I am using within my lab. Thank you so much for your help!

closed 2024-06-27 by EliasDubroff

BenediktBurger, 2024-06-27:
By the way, you can use anything in the pymeasure measurement routines (the GUI, etc.), not only pymeasure instruments. Therefore you can use the nidaqmx package or similar for NI access.

EliasDubroff, 2024-06-27:
What I was thinking about doing was using the nidaqmx package within the methods of my created DAQ class. I'm curious what you meant by that though, could you elaborate? I would happily do this in a different way if it would be easier or more efficient.