pymeasure/pymeasure #1155: How to execute a procedure manually

SengerM, 2024-08-20:
I am following [the tutorial](https://pymeasure.readthedocs.io/en/latest/tutorial/procedure.html) where it is explained how to create a simple procedure and run it using a `Worker`. Though this is really nice, in some cases it is handy to just run a procedure in the main thread, which I don't know how to do. Consider this example:

```python
from pymeasure.experiment import Procedure

class Whatever(Procedure):
	DATA_COLUMNS = ['foo','bar']
	
	def startup(self):
		print('Hi, I am starting!')
	
	def execute(self):
		print('Hi, I am executing!')
		self.emit('results', dict(foo=1, bar=2))
	
	def shutdown(self):
		print('Bye, chau, tschuss!')

if __name__ == '__main__':
	p = Whatever()
	
	p.startup()
	p.execute()
	p.shutdown()
```
which produces
```
$ python deleteme.py 
Hi, I am starting!
Hi, I am executing!
Traceback (most recent call last):
  File "/home/me/Desktop/deleteme.py", line 20, in <module>
    p.execute()
  File "/home/me/Desktop/deleteme.py", line 11, in execute
    self.emit('results', dict(foo=1, bar=2))
  File "/home/me/code/pymeasure/pymeasure/experiment/procedure.py", line 282, in emit
    raise NotImplementedError('should be monkey patched by a worker')
NotImplementedError: should be monkey patched by a worker
```

 So I have two questions regarding this:

1. How would I run my procedure manually?
2. How would I capture the data sent by `self.emit` in this case?

CasperSchippers, 2024-08-22:
Hey @SengerM 
Hmmm, that is a really interesting question. I am not sure there is at this moment a good solution for it actually (at least, not that I can think of at this moment), but I can imagine there are use cases for this indeed. Maybe a context manager mode could be a nice approach indeed, although I don't have the implications sharp from the top of my head.

To get a bit better understanding of what you want to achieve: what is the exact reason you want to run it in the main thread, rather than in a worker thread?

CasperSchippers, 2024-08-25:
Thanks for expanding on your uses;
I understand the first use.

I'm not sure if I fully get why you'd want to do that; at this moment, i'm inclined to think that you'd better achieve that using 'ordinary' methods/functions, or class if you perse want something like a context manager. If you use a procedure, you run into the issue that emitting the data will be handled by the inner procedure, such that a single file is written for every MeasurePoint, without any information on for which temperature of voltage this measurement point is taken. If you do want a seperate file for every temperature/voltage pare, I'd rather suggest using the [`SequencerWidget`](https://pymeasure.readthedocs.io/en/latest/tutorial/graphical.html#using-the-sequencer) with a single procedure.

Otherwise, if you want to have one file contaiing all temperature/voltage pairs, I'd rather suggest something like this (as this will also reduce the amount of overhead of starting and stopping the communication with the respective instruments):
```python
class PointsVsVoltageVsTemperature(Procedure):
    def measure_point(self):
        ...

    def execute(self):
        for T in temperatures:
            ...  # define temperatures and voltages
            for (T, V) in product(temperatures, voltages):  # product from itertools
                self.measure_point()
```

SengerM, 2024-08-28:
@CasperSchippers indeed, for the case number 2, it is an overkill to measure a single point in that way, also super inefficient. However, `MeasurePoint` could actually be a quite complex procedure not creating just a single "row" of data, but a lot of data. Then the other procedures just call it so there is no need to rewrite any code. Additionally, and if needed, the merging of the individual files into a larger file could be done in the `__exit__` or in the `shutdown` method.

What I did a lot in the past, not with pyMeasure but with my own framework (which is of course non-standard and only used by me), is very similar to what I described above. The `MeasurePoint` was, for example, a 1D scan with a pulsed laser on a sample, with all variables fixed. Such a scan produced some gigabytes of data. Then I nested this inside a `Measure2D` which would do multiple 1D scans in `x`, each at a different `y`. Finally, I nested this 2D scan inside a `Measure2DSweepingVoltage` that was, as the name suggest, repeating this for multiple voltages. In this way, the extension of the original "procedure" was close to trivial, no code duplication, and the data analysis methods were implemented in the same way, because the data format automatically follows a nested tree structure, which once more makes it very easy.

CasperSchippers, 2024-09-04:
The use case you describe sounds like a perfect use case for the sequencer widget.

I'm not a big fan of rewriting the datafiles, especially if there gigabytes large, this quite a bit of time and seems error prone. Plus, if you abort your measurement before (or maybe even worse, during) the rewriting, you do not have a valid datafile (in contrast to the standard architecture).

If you do want to manually nest the procedures, you could also monkeypatch the emit method with a custom method that writes the file directly the way you want to. But I do think this (both approaches actually) results in a lot of glue-code, so I think I'm still more in favour of the sequencer widget or additional methods/classes (which requires only a little bit of overhead I expect).

That being said, I would recommend having a look at the sequencer widget (that is, unless you cannot use a user interface, in which case maybe the ManagedConsole can be used, or we need to look for a different solution)

PS Sorry for the slow (even slower than normal) reply.