pymeasure/pymeasure #1170: Error trying to follow the contributing guide for PyMeasure

OrionSmedley, 2024-10-07:
Following the [instructions](https://pymeasure.readthedocs.io/en/latest/dev/contribute.html), I went into the repository folder, and ran
`pip install -e .`
But got the following message
```
ERROR: File "setup.py" not found. Directory cannot be installed in editable mode
(A "pyproject.toml" file was found, but editable mode currently requires a setup.py based build.)
```

Any help?

BenediktBurger, 2024-10-07:
What pip version do you use? probably an old one.
Try `pip install pip --upgrade`

OrionSmedley, 2024-10-08:
Tried that. 

Now says:
```
(base) C:\Users\Spin Doctor\Documents\Github\pymeasure>pip install -e .
Obtaining file:///C:/Users/Spin%20Doctor/Documents/Github/pymeasure
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... error
  error: subprocess-exited-with-error

  × Getting requirements to build editable did not run successfully.
  │ exit code: 1
  ╰─> [57 lines of output]
      WARNING setuptools_scm.run_cmd command git missing: [WinError 2] The system cannot find the file specified
      ERROR setuptools_scm._get_version_impl command git not found while parsing the scm, using fallbacks
      Traceback (most recent call last):
        File "C:\Users\Spin Doctor\AppData\Local\Temp\pip-build-env-2_13zq0h\overlay\Lib\site-packages\setuptools_scm\_get_version_impl.py", line 40, in parse_scm_version
          return _entrypoints.version_from_entrypoint(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Spin Doctor\AppData\Local\Temp\pip-build-env-2_13zq0h\overlay\Lib\site-packages\setuptools_scm\_entrypoints.py", line 57, in version_from_entrypoint
          maybe_version: version.ScmVersion | None = fn(root, config=config)
                                                     ^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Spin Doctor\AppData\Local\Temp\pip-build-env-2_13zq0h\overlay\Lib\site-packages\setuptools_scm\git.py", line 212, in parse
          _require_command("git")
        File "C:\Users\Spin Doctor\AppData\Local\Temp\pip-build-env-2_13zq0h\overlay\Lib\site-packages\setuptools_scm\_run_cmd.py", line 214, in require_command
          raise CommandNotFoundError(name)
      setuptools_scm._run_cmd.CommandNotFoundError: git
      Traceback (most recent call last):
        File "C:\Users\Spin Doctor\miniconda3\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 353, in <module>
          main()
        File "C:\Users\Spin Doctor\miniconda3\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Spin Doctor\miniconda3\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 132, in get_requires_for_build_editable
          return hook(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Spin Doctor\AppData\Local\Temp\pip-build-env-2_13zq0h\overlay\Lib\site-packages\setuptools\build_meta.py", line 464, in get_requires_for_build_editable
          return self.get_requires_for_build_wheel(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Spin Doctor\AppData\Local\Temp\pip-build-env-2_13zq0h\overlay\Lib\site-packages\setuptools\build_meta.py", line 332, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=[])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Spin Doctor\AppData\Local\Temp\pip-build-env-2_13zq0h\overlay\Lib\site-packages\setuptools\build_meta.py", line 302, in _get_build_requires
          self.run_setup()
        File "C:\Users\Spin Doctor\AppData\Local\Temp\pip-build-env-2_13zq0h\overlay\Lib\site-packages\setuptools\build_meta.py", line 318, in run_setup
          exec(code, locals())
        File "<string>", line 1, in <module>
        File "C:\Users\Spin Doctor\AppData\Local\Temp\pip-build-env-2_13zq0h\overlay\Lib\site-packages\setuptools\__init__.py", line 117, in setup
          return distutils.core.setup(**attrs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Spin Doctor\AppData\Local\Temp\pip-build-env-2_13zq0h\overlay\Lib\site-packages\setuptools\_distutils\core.py", line 145, in setup
          _setup_distribution = dist = klass(attrs)
                                       ^^^^^^^^^^^^
        File "C:\Users\Spin Doctor\AppData\Local\Temp\pip-build-env-2_13zq0h\overlay\Lib\site-packages\setuptools\dist.py", line 283, in __init__
          _Distribution.__init__(self, dist_attrs)
        File "C:\Users\Spin Doctor\AppData\Local\Temp\pip-build-env-2_13zq0h\overlay\Lib\site-packages\setuptools\_distutils\dist.py", line 279, in __init__
          self.finalize_options()
        File "C:\Users\Spin Doctor\AppData\Local\Temp\pip-build-env-2_13zq0h\overlay\Lib\site-packages\setuptools\dist.py", line 635, in finalize_options
          ep(self)
        File "C:\Users\Spin Doctor\AppData\Local\Temp\pip-build-env-2_13zq0h\overlay\Lib\site-packages\setuptools_scm\_integration\setuptools.py", line 123, in infer_version
          _assign_version(dist, config)
        File "C:\Users\Spin Doctor\AppData\Local\Temp\pip-build-env-2_13zq0h\overlay\Lib\site-packages\setuptools_scm\_integration\setuptools.py", line 58, in _assign_version
          _version_missing(config)
        File "C:\Users\Spin Doctor\AppData\Local\Temp\pip-build-env-2_13zq0h\overlay\Lib\site-packages\setuptools_scm\_get_version_impl.py", line 117, in _version_missing
          raise LookupError(
      LookupError: setuptools-scm was unable to detect version for C:\Users\Spin Doctor\Documents\Github\pymeasure.

[... cut ...] work.

      For example, if you're using pip, instead of https://github.com/user/proj/archive/master.zip use git+https://github.com/user/proj.git#egg=proj
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build editable did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

BenediktBurger, 2024-10-09:
Did you download the Repository files or did you clone it into your own local Repository?

The error message indicates the former one, while you have to clone it.

OrionSmedley, 2024-10-09:
I cloned it with github desktop.

OrionSmedley, 2024-10-11:
What branch should I be in? @BenediktBurger 
`pip install -e .` fails in both main and develop.

Tried on another computer. and got the same problem.

Following the directions, I forked the main repository,
https://github.com/OrionSmedley/pymeasure
then cloned it with GitHub desktop.

Do you see any issues with this? Maybe the version that I forked is funky?

OrionSmedley, 2024-10-11:
ok, I can get the old branch, instruments-features, to work with Python 3.9.12

OrionSmedley, 2024-10-12:
More info:
Python: 3.12.7
Conda: 24.7.1
Pip: 24.2

OrionSmedley, 2024-10-14:
Aha, found out the secret. Maybe we should add this to the documentation.

pip cannot install in editable mode, if you just use github desktop.
You need an accessible version of git that pip can call.

GitHub Desktop does install git, but it's burried, and not on the path. 
The following fixes the problem
`set PATH="C:\Users\<USERNAME>\AppData\Local\GitHubDesktop\app-<#######>\resources\app\git\cmd";%PATH%`
Then `pip install -e .` works like a charm

BenediktBurger, 2024-10-15:
Danke für den Hinweis

closed 2024-10-17 by BenediktBurger