pymeasure/pymeasure #1177: KeithleyDAQ6510 instrument not available

OliviaSeidel, 2024-10-24:
Hello, It looks like the instrument KeithleyDAQ6510 is not available in the latest version of pymeasure, although it is listed in the documentation. 

In keithly/_init_.py there are the following instruments, and the 6510 is not there:

'''
from .keithley2000 import Keithley2000
from .keithley2260B import Keithley2260B
from .keithley2306 import Keithley2306
from .keithley2400 import Keithley2400
from .keithley2450 import Keithley2450
from .keithley2600 import Keithley2600
from .keithley2700 import Keithley2700
from .keithley2750 import Keithley2750
from .keithley6221 import Keithley6221
from .keithley6517b import Keithley6517B
from .keithley2200 import Keithley2200
from .keithleyDMM6500 import KeithleyDMM6500
from .keithley2182 import Keithley2182
'''

BenediktBurger, 2024-10-25:
Hello Olivia,

You were looking at the _latest_ documentation, which reflects the state of the master branch here on github, which includes the DAQ6510 (as a file and in the init).
If you change to the _stable_ [documentation](https://pymeasure.readthedocs.io/en/stable/) (also visible on the bottom right), you see, that the device is not included.
The reason is, this device has been added to pymeasure since the last release.

If you want to use this device until we release a new pymeasure version, you have to use git to clone this repository and install it.
The [documentation](https://pymeasure.readthedocs.io/en/latest/dev/contribute.html#using-the-development-version) tells how to do it.
If you have questions, don't hesitate to ask.

Teslafly, 2025-06-06:
I think this can be closed. The KeithleyDAQ6510 is now included in the release version. I have been able to use it. And am now working on some improvements.