pymeasure/pymeasure #1178: Move from `setup.cfg` fully to `pyproject.toml`

RobertoRoos, 2024-10-25:
A thought I was having: are we interested in replacing `setup.cfg` entirely by `pyproject.toml`? We already have the latter, but it's used for some tertiary config elements.

In my understanding `pyproject.toml` is the latest and preferred method of package definition.

I should add however that it looks like `setup.cfg` is not being deprecated soon: https://stackoverflow.com/questions/44878600/is-setup-cfg-deprecated

BenediktBurger, 2024-10-25:
The idea was, to keep backwards compatibility (with older pip versions), as some systems have old python versions etc.

According to the [pip documentation](https://pip.pypa.io/en/stable/reference/build-system/pyproject-toml/), version 10.0 (released April 14, 2018) started to support pyproject.toml.
Python 3.7.0 was released June 27, 2018, and included pip 10.0.1.

As our minimum version is 3.8, which is already end of life, we can change to a `pyproject.toml`, I think.
What do you think, @bilderbuchi ?

As Flake8 does not [support configuration](https://flake8.pycqa.org/en/latest/user/configuration.html) in pyproject.toml, only in setup.cfg or .flake8, I'd move the configuration to a `.flake8` file.

bilderbuchi, 2024-10-28:
> The idea was, to keep backwards compatibility (with older pip versions), as some systems have old python versions etc.

Yes, indeed, that was the reasoning, way back when we made the transition from setup.py to setup.cfg. 
Yes, I agree, it now seems safe to fully move to pyproject.toml now.

> As our minimum version is 3.8, which is already end of life

Ha, I missed that date passing by. We should add an issue to track bumping our minimum to 3.9 and modernise the codebase with any new features thus available.

> I'd move the configuration to a .flake8 file.

Agreed, that makes sense as it allows dropping the setup.cfg file completely, which should enhance clarity for people browsing the codebase (it's immediately clear where the Python project metadata reside). Flake8 is weird like that. :-P

RobertoRoos, 2024-10-28:
Nice. I'll try to prepare a branch for `pyproject.toml`. Let's do the step away from Python 3.8 in a separate PR then.

PR #1182 (2024-10-28): Removed setup.cfg, replaced by pyproject.toml
Resolves #1178
closed 2024-10-28 by BenediktBurger