pymeasure/pymeasure #1179: Pin development tools for the package

RobertoRoos, 2024-10-25:
From what I see there is no quick way to get the additional Python packages installed to start developing `pymeasure`, like `pytest`, `black`, `flake8`, `sphinx`, etc. The errors are typically clear when required tools are missing, so installing them manually is no large concern. But then there is still the matter of versions.

A solution could be option requirement groups, such that when starting development the following installs everything to check, test and build docs:
```
pip install -e .[dev,test,doc]
``` 

Poetry also has optional groups for that, `poetry install --with dev`, but I don't know if we'd want to switch backends.

RobertoRoos, 2024-10-25:
Whoops, I had completely missed exactly this does already exists, see https://github.com/pymeasure/pymeasure/blob/master/setup.cfg#L40

So `pip install -e .[tests]` is available, installing pytest and plugins.

Then I guess the suggestion would be add more, also for linters and for sphinx.

BenediktBurger, 2024-10-25:
A group for the documentation is fine.
We do provide linter configuration for different linters, but do not require one or the other, therefore, I'd prefer not to include it.
Feel free to open such a PR.

RobertoRoos, 2024-10-25:
> We do provide linter configuration for different linters, but do not require one or the other, therefore, I'd prefer not to include it.

Doesn't `flake8` run on CI for each pull request? Then it makes sense to pin the same Flake8 version somewhere such that contributors could exactly reproduce the CI step themselves, allow us to fix issues without having to push code first.

BenediktBurger, 2024-10-25:
Yes, CI runs flake8, but you can use any other linter, which produces good Python code (black, ruff etc.).

PR #1180 (2024-10-25): Added an install group for building documentation
Resolves #1179
closed 2024-10-25 by BenediktBurger