pymeasure/pymeasure #1190: Confirmation: AnritsuMS2090A supports SCPI commands

icchalmers, 2024-11-13:
Can confirm the Anritsu MS2090A supports the basic SCPI commands from `SCPIMixin`. I couldn't find a common thread to report this, so apologies if an issue isn't the best place.

https://github.com/pymeasure/pymeasure/blob/e778b2f8338df81d7572caf3e9280e43a1624e65/pymeasure/instruments/anritsu/anritsuMS2090A.py#L36

Testing using:

```python
from pymeasure.instruments.anritsu import AnritsuMS2090A
dev = AnritsuMS2090A("TCPIP0::10.1.1.141::9001::SOCKET", read_termination="\n", write_termination="\n")
print(dev.complete)
print(dev.status)
print(dev.options)
print(dev.id)
print(dev.next_error)
```

BenediktBurger, 2024-11-13:
It's fine to open an issue.

Would you mind open a Pull request changing the SCPIUnknownMixin to SCPIMixin (in the class definition and at the imports)?

Thanks!

icchalmers, 2024-11-14:
Probably should have just done that in the first place. Thanks to you and the other maintainers/contributors for the good work!

PR #1191 (2024-11-18): AnritsuMS2090A: Confirm SCPIMixin
Closes #1190
closed 2024-11-18 by BenediktBurger