pymeasure/pymeasure #1200: Error -113 (Undefined SCPI header) when using the tutorial example with Keithley 2450

dsimat, 2024-12-19:
Hello,

I am trying to use pymeasure with Keithley 2450 and the [measurement script](https://pymeasure.readthedocs.io/en/latest/tutorial/procedure.html) provided in the tutorial does not work. The script is copied below for convenience:

```
# Import necessary packages
from pymeasure.instruments.keithley import Keithley2450
import numpy as np
import pandas as pd
from time import sleep

# Set the input parameters
data_points = 50
averages = 10
max_current = 0.001
min_current = -max_current

# Set source_current and measure_voltage parameters
current_range = 10e-3  # in Amps
compliance_voltage = 10  # in Volts
measure_nplc = 0.1  # Number of power line cycles
voltage_range = 1  # in VOlts

# Connect and configure the instrument
sourcemeter = Keithley2450("GPIB::7::INSTR")
sourcemeter.reset()
sourcemeter.use_front_terminals()
sourcemeter.apply_current(current_range, compliance_voltage)
sourcemeter.measure_voltage(measure_nplc, voltage_range)
sleep(0.1)  # wait here to give the instrument time to react
sourcemeter.stop_buffer()
sourcemeter.disable_buffer()

# Allocate arrays to store the measurement results
currents = np.linspace(min_current, max_current, num=data_points)
voltages = np.zeros_like(currents)
voltage_stds = np.zeros_like(currents)

sourcemeter.enable_source()

# Loop through each current point, measure and record the voltage
for i in range(data_points):
    sourcemeter.config_buffer(averages)
    sourcemeter.source_current = currents[i]
    sourcemeter.start_buffer()
    sourcemeter.wait_for_buffer()
    # Record the average and standard deviation
    voltages[i] = sourcemeter.means[0]
    sleep(1.0)
    voltage_stds[i] = sourcemeter.standard_devs[0]

# Save the data columns in a CSV file
data = pd.DataFrame({
    'Current (A)': currents,
    'Voltage (V)': voltages,
    'Voltage Std (V)': voltage_stds,
})
data.to_csv('example.csv')

sourcemeter.shutdown()
```

The only changes I made was to replace "Keithley2400" with "Keithley2450" and to the GPIB port of the instrument.

The issue is that the wait_for_buffer() function leads to a -113 error (Undefined SCPI header). Strangely, this script did work when I first tested pymeasure back in September. So, I cannot find what changed since then. I also attach two images that show the error.

[image]

[image]

dsimat, 2024-12-19:
Update: the `.disable_buffer()` method in the script seems to trigger the error 113.

dsimat, 2024-12-19:
Update: the code in `buffer.py` seems to be written for the Keithley 2400. It doesn't work for Keithley 2450. However, if I go to Menu --> System --> Settings and set the Command Set to "SCPI 2400", then all the pieces of code work. They just seem to all be written for the 2400 model.

BenediktBurger, 2024-12-21:
Thanks for investigating. 

Would you mind adding a relevant documentation code to the device docstring?

dsimat, 2024-12-25:
Sure. But should I edit the device docstring (i.e. the docstring of `Keithley2450.py`)? I guess it is the doctring of `buffer.py` that needs to be edited. I can just add a brief note that it doesn't work with Keithley 2450 unless the command set is set to SCPI 2400.

BenediktBurger, 2025-01-16:
The  Keithley buffer (from another module) is used, right?

In this case add the note to the device itself, please.

Grandpa-G, 2025-02-07:
Will this get fixed?

Bigger question, what else for the 2450 doesn't work? All the buffer methods?

dsimat, 2025-02-07:
Sorry, I'll update the docstring this weekend. Yes, the buffer methods work only for 2400. If you look at 2450's manual you'll see that the syntax of the buffer-handling SCPI commands is different from the syntax used in 2400. That's the crux of the problem. Hence, we would need a different buffer handling python script in order to use the native buffer-handling commands of 2450.

dsimat, 2025-02-07:
See [here](https://download.tek.com/manual/2450-901-01_D_May_2015_Ref.pdf) for details. The differences between 2400 and 2450 code are in the appendix.

Grandpa-G, 2025-02-07:
Thanks, I guess.

dsimat, 2025-02-07:
Just to clarify, you can still use the 2450. You just need to set its SCPI command to emulate that of 2400, as described earlier in this thread. Then you can load the Keithley 2400 module using pymeasure and use the 2450 as if it was the 2400 model.

dsimat, 2025-02-13:
I have sent pull request [#1238](https://github.com/pymeasure/pymeasure/pull/1238) with the edited docstring. Happy to edit/add more as needed.

PR #1238 (2025-02-26): Updated Keithley2450's docstring to inform that buffer handling is currently unsupported
I have added a note in the docstring of class Keithley2450 that informs that buffer handling for the 2450 is currently unsupported but that the instrument can be used in emulation mode as a 2400 model.

Closes #1200
closed 2025-02-26 by BenediktBurger