pymeasure/pymeasure #1203: Incorrect command in documentation to clone pymeasure

dazz100, 2024-12-31:
On this web page:
[https://pymeasure.readthedocs.io/en/latest/dev/contribute.html](url)

Th instructions include an error.  The instructions say:
```
Clone your fork of PyMeasure your-github-username/pymeasure. In the following terminal commands replace your desired path and GitHub username.

cd /path/for/code
git clone https://github.com/your-github-username/pymeasure.git
```
The instructions should say:

```
Clone your fork of PyMeasure pymeasure/pymeasure. In the following terminal commands replace your desired path and GitHub username.

cd /path/for/code
git clone https://github.com/pymeasure/pymeasure.git
```

This command succeeds  and gives the following output:

```
~/git/pymeasure$ git clone https://github.com/pymeasure/pymeasure.git
Cloning into 'pymeasure'...
remote: Enumerating objects: 29713, done.
remote: Counting objects: 100% (11689/11689), done.
remote: Compressing objects: 100% (1086/1086), done.
remote: Total 29713 (delta 11211), reused 10607 (delta 10603), pack-reused 18024 (from 3)
Receiving objects: 100% (29713/29713), 8.98 MiB | 5.60 MiB/s, done.
Resolving deltas: 100% (22290/22290), done.

```

Please correct the instructions.

BenediktBurger, 2025-01-02:
You skipped a previous step a few lines above your quote:

> In order to add your feature, you need to first fork PyMeasure. This will create a copy of the repository under your GitHub account.

The word fork is a link to an explanation.

If you did this, you have to clone your fork and not the original pymeasure Repository.

Would you mind revisit the documentation and look, how it could be made clearer?

Feel free to open such a pull request or tell it here.

dazz100, 2025-01-02:
Hi
The documentation would be a lot clearer if I knew more about git.  
I ended up deleting everything and forked PyMeasure to create a new repository on my GitHub account. I have linked my local version to my forked PyMeasure.  

That's probably what I should have done in the first place, but I am still on the bottom of the git learning curve.

BenediktBurger, 2025-01-03:
Should we keep the documentation as is, or should we modify it to be more clear for beginners?

dazz100, 2025-01-04:
This YT video is the best I have seen for someone who knows nothing about git.
[https://www.youtube.com/watch?v=mJ-qvsxPHpY&list=PL-WKO2ElUnsKI790FYjp_3IVz7TwPKQAR](url)
Maybe just add a link to this video.

BenediktBurger, 2025-01-16:
I'm not that happy, linking to a YouTube video, as the link may break etc. I prefer to make the text a little clearer and let the people do a online research by themselves, if they need more background information.

dazz100, 2025-01-16:
OK so maybe add a section at the top that is basically a checklist of steps needed to build a development environment.  
and at the end, some simple tests/checks to confirm the environment is correctly setup.

The Checklist section might look something like:

If you haven't used git or git hub before, the following checklist summarises the steps to build the development environment:
1.  Setup Anaconda
2. Create a Github account.
3. Setup ssh credentials with Github
4. Uninstall any existing versions of pymeasure
5. Install pymeasure, configured for editing
6. ....

Each of these steps is described below and in the links.