pymeasure/pymeasure #1212: PyQt6 compatibility issues

Benbriel, 2025-01-09:
I know this is not a bug, since PyMeasure doesn't enforce any specific Qt library, but IMO, PyMeasure could have better Qt version management. PyMeasure imports Qt objects via `pyqtgraph.Qt` ([pymeasure/display/Qt.py](https://github.com/pymeasure/pymeasure/blob/master/pymeasure/display/Qt.py)) without a way to manage differences between PyQt5 and PyQt6 (or PySide2 and PySide6). This leads to compatibility issues with PyQt6, mainly because of the API changes introduced.

### Examples:

- **`FileInputWidget`** ([pymeasure/display/widgets/fileinput_widget.py#L55](https://github.com/pymeasure/pymeasure/blob/master/pymeasure/display/widgets/fileinput_widget.py#L55)):  
   ```python
   self.writefile_toggle.setLayoutDirection(QtCore.Qt.RightToLeft)
   ```
   This works in PyQt5 but fails in PyQt6, where the correct call is:  
   ```python
   QtCore.Qt.LayoutDirection.RightToLeft
   ```

- **`PlaceholderCompleter`** ([pymeasure/display/widgets/filename_widget.py#L77](https://github.com/pymeasure/pymeasure/blob/master/pymeasure/display/widgets/filename_widget.py#L77)):  
   ```python
   self.setCaseSensitivity(QtCore.Qt.CaseInsensitive)
   ```
   In PyQt6, the correct call is:  
   ```python
   QtCore.Qt.CaseSensitivity.CaseInsensitive
   ```

- [pymeasure/display/widgets/filename_widget.py#L78](https://github.com/pymeasure/pymeasure/blob/master/pymeasure/display/widgets/filename_widget.py#L78)

- [pymeasure/display/widgets/filename_widget.py#L123](https://github.com/pymeasure/pymeasure/blob/master/pymeasure/display/widgets/filename_widget.py#L123)

- [pymeasure/display/widgets/filename_widget.py#L125](https://github.com/pymeasure/pymeasure/blob/master/pymeasure/display/widgets/filename_widget.py#L125)

- [pymeasure/display/widgets/filename_widget.py#L127](https://github.com/pymeasure/pymeasure/blob/master/pymeasure/display/widgets/filename_widget.py#L127)

References to `QtCore.Signal()` are correctly handled by pyqtgraph, pointing to `PyQtX.QtCore.pyqtSignal()` when using PyQtX. 

### Proposed solutions:

- Introduce a version manager within PyMeasure, like [QtPy](https://github.com/spyder-ide/qtpy) or [Qt.py](https://github.com/mottosso/Qt.py), to handle Qt differences 
- Stick to one Qt version consistently across modules (PyQt6 syntax seems to work with `pyqtgraph~0.13` for both PyQt5 and PyQt6)
- Document and enforce compatible Qt versions to ensure consistent behavior across setups.
- Manage API differences explicitly in [pymeasure/display/Qt.py](https://github.com/pymeasure/pymeasure/blob/master/pymeasure/display/Qt.py)

I apologize if I got something wrong. Let me know if you think there's a better way to approach this, or if it's not needed at all. I'm happy to keep contributing!

BenediktBurger, 2025-01-12:
Thanks for raising this issue.

@CasperSchippers could you take a look, please?

I'm not an expert on pymeasure GUI, but generally speaking, using a library like qtpy consistently seems a good idea.
Maybe we could add a few tests for the different qt libraries.

CasperSchippers, 2025-01-13:
Thanks for bringing this up. Maybe indeed it would be a good idea introduce something like QtPy or Qt.Py; people are maintaining this layer anyways, so why reinvent the wheel....

I'm not sure if there generally is a preference to either one? It seems both are well maintained and QtPy I think is used more often?

BenediktBurger, 2025-01-13:
I'm using qtpy in the lab and it works quite well. From Qt.Py I don't know anything.

Benbriel, 2025-01-14:
Qt.py doesn't have pyqt6 support (yet), so I personally prefer qtpy. It wouldn't be much harder than replacing imports in [pymeasure/display/Qt.py](https://github.com/pymeasure/pymeasure/blob/master/pymeasure/display/Qt.py):

```
from qtpy import QtCore, QtGui, QtWidgets
from qtpy.uic import loadUiType
```
I'm unaware of how pyqtgraph and qtpy interact, however.

seb5g, 2025-01-14:
I'm using qtpy within pymodaq and it has complete qt6 compatibility. Pyqtgraph build their own layer. I'm using both together without any issue

CasperSchippers, 2025-01-14:
Then let's settle on qtpy.

hvegh, 2025-04-21:
> Qt.py doesn't have pyqt6 support (yet), so I personally prefer qtpy. It wouldn't be much harder than replacing imports in [...]

I can confirm this change works with the  [gui.py](/pymeasure/pymeasure/blob/master/examples/Basic/gui.py) example

Benbriel, 2025-04-22:
Started  #1285 to handle this

closed 2025-05-24 by Benbriel