pymeasure/pymeasure #1216: Release v0.15.0 on conda-forge

BenediktBurger, 2025-01-15:
@bilderbuchi would you mind update the conda-forge feedstock for v0.15.0?

bilderbuchi, 2025-01-16:
I think the conda forge bot will even automatically create a PR at https://github.com/conda-forge/pymeasure-feedstock/pulls, so I'll wait for that, first. 
I estimate that we will only have to cross-check the dependency versions at recipe/meta.yaml (if anything changed, and maybe if we have to adapt for the new optional dependencies for docs).

bilderbuchi, 2025-01-16:
In any case, I added you to the maintainer list, should have done that a long time ago: https://github.com/conda-forge/pymeasure-feedstock/issues/41

BenediktBurger, 2025-01-16:
Cool, I did not know, that a Conda bot checks for new versions or did you do that in the recipe?

bilderbuchi, 2025-01-16:
> Cool, I did not know, that a Conda bot checks for new versions or did you do that in the recipe?

CF added that capability at some point in the (recentish) past.

BenediktBurger, 2025-01-25:
It's released, so we can close this issue, can't we?

Or do you want to keep it open, until we resolved the Conda version discussion?