pymeasure/pymeasure #1258: Agilent33220A Offset Range Wrong

Patschke, 2025-03-26:
I'm using pymeasure to talk to an Agilent33220A AWG. I've noticed that the parameter range for the offset is too strict (and perhaps also for some other parameters, I did not test that in detail). It's currently hard coded to -4.995 to +4.995:

https://github.com/pymeasure/pymeasure/blob/c9e806c7c30d4c9ea48424c3bc8d9827570ff4e1/pymeasure/instruments/agilent/agilent33220A.py#L124

However, the actual  range depends on the output termination. You can change that (default is 50 Ohm) in the Utility->Output Setup menu. 

[image]

The voltage range can go up to -9.95...+9.95 V for larger load resistances: 

[image]

Via pymeasure it's still only possible to set values in the 50 Ohm range of 5 V.

BenediktBurger, 2025-03-28:
Thanks for submitting this bug report.
Would you mind adjusting the limits?

Could you please check after the change, what the device (in 50 Ohm resistance) does, if you exceed +- 5V? Does pymeasure raise an exception?

Thanks!

BenediktBurger, 2025-04-01:
> Given that the current policies for the frequency already is to fail silently (setting a frequency above 200 kHz while in Ramp mode shows the same error on the device as setting an offset Voltage above 5 V in 50 Ohm termination) I could see simply raising all the limits?

Instead of raising the limits, you can remove the limits altogether, as they depend on the configuration which we do not know at all. Also the docstring should state, that the limits depend on the current configuration.

Thanks for showing the technical details.

closed 2025-08-07 by BenediktBurger