pymeasure/pymeasure #1261: pytest fails with numerical values in QtCore.QtLocale() call

dl8dtl, 2025-03-28:
When running pytest here, I get complaints:
```
______________________________________________________ ERROR collecting tests/display/test_inputs.py _______________________________________________________
tests/display/test_inputs.py:159: in <module>
    class TestScientificInput:
tests/display/test_inputs.py:244: in TestScientificInput
    [QtCore.QLocale(31, 7, 224), "."],  # UK locale for period
E   TypeError: 'PySide6.QtCore.QLocale.__init__' called with wrong argument types:
E     PySide6.QtCore.QLocale.__init__(int, int, int)
E   Supported signatures:
E     PySide6.QtCore.QLocale.__init__()
E     PySide6.QtCore.QLocale.__init__(PySide6.QtCore.QLocale.Language, PySide6.QtCore.QLocale.Script = Instance(QLocale.Script.AnyScript), PySide6.QtCore.QLocale.Country = Instance(QLocale.Country.AnyTerritory))
E     PySide6.QtCore.QLocale.__init__(PySide6.QtCore.QLocale.Language, PySide6.QtCore.QLocale.Country)
E     PySide6.QtCore.QLocale.__init__(str)
E     PySide6.QtCore.QLocale.__init__(Union[PySide6.QtCore.QLocale, PySide6.QtCore.QLocale.Language])
================================================================= short test summary info ==================================================================
ERROR tests/display/test_inputs.py - TypeError: 'PySide6.QtCore.QLocale.__init__' called with wrong argument types:
```
Why not using symbolic (and thus correctly typed) values there?

The following patch makes it work for me:
```
diff --git a/tests/display/test_inputs.py b/tests/display/test_inputs.py
index cc5c064a..a768c48a 100644
--- a/tests/display/test_inputs.py
+++ b/tests/display/test_inputs.py
@@ -241,8 +241,8 @@ class TestScientificInput:
             p.assert_called_once_with(5.0)
 
     @pytest.mark.parametrize("locale, decimalSep", [
-        [QtCore.QLocale(31, 7, 224), "."],  # UK locale for period
-        [QtCore.QLocale(30, 7, 151), ","],  # NL locale for comma
+        [QtCore.QLocale(QtCore.QLocale.English, QtCore.QLocale.LatinScript, QtCore.QLocale.UnitedKingdom), "."],
+        [QtCore.QLocale(QtCore.QLocale.Dutch, QtCore.QLocale.LatinScript, QtCore.QLocale.Netherlands), ","],
     ])
     def test_locale_settings(self, qtbot, locale, decimalSep):
         assert locale.decimalPoint() == decimalSep
```
Besides, it obviates the need for the former comment.

BenediktBurger, 2025-03-28:
@CasperSchippers maybe you could have a look at that?

CasperSchippers, 2025-03-29:
Hmmm, yeah, that is a lot better actually 😅. Nice, thanks for the find

closed 2025-03-29 by CasperSchippers