pymeasure/pymeasure #1298: Missing color bar scale  in the image plotting functionality

PkRPintu, 2025-05-09:
Hello,

I am new to Pymeasure and would like to use it for live image scanning while grabbing data from various lock-ins. I have three questions if anyone can kindly help me. I really appreciate any help you can offer.

1. How can I add a color bar to the image scan display that changes dynamically depending on the Z axis values? (If the colour bar can be interactive, that would also be a great feature)
2. How to change the colormap from viridis to something else, probably a diverging colour map like ‘bwr’?
3. Also, there is a managed docked window for displaying multiple plots. Can we have managed docked image windows so that we can display multiple image plots at once?

I think these are very handy features for the image scan experiments and will be helpful for everyone.

Cheers

CasperSchippers, 2025-06-04:
Hey,

These are good, but difficult questions, to which I don't have a good awnser to all of them.

> 1. How can I add a color bar to the image scan display that changes dynamically depending on the Z axis values? (If the colour bar can be interactive, that would also be a great feature)

While I think it should be possible, I didn't manage to add one just yet... I do agree that this would be very, very useful!
By the way, what do you mean with interactive?

> 2. How to change the colormap from viridis to something else, probably a diverging colour map like ‘bwr’?

To change the colormap, you currently need to alter a single line in `curves.py`, where it specifies the colormap: https://github.com/pymeasure/pymeasure/blob/13a173fc465b0b0b5daabff2e4fe55ae4d2ff94a/pymeasure/display/curves.py#L87
Pyqtgraph does include colormaps of colorcet (https://colorcet.holoviz.org/, which I think are generally quite good, so for most purposes I expect this would suffice), so it includes a set of diverging colormaps, such as `CET-D1` for bwr, or `CET-D4` for bkr, or `CET-D6` for bky (these ones have the advantage that zero also stands out from the background of the plotting window), or (one which I particullarly like as it is isoluminant to prevent misguided interpretations) `CET-D12`/`CET-D11`.

For a diverging colormap, you may also want to adjust the edges of the colormap, such that zero is the center of the map (i.e. white, or gray, or black, depending on the exact map that is used). For this you can e.g. update the zmin and zmax values after these lines: https://github.com/pymeasure/pymeasure/blob/13a173fc465b0b0b5daabff2e4fe55ae4d2ff94a/pymeasure/display/curves.py#L104-L105
For example by adding some lines such as:
``` python
        zmax = max(abs(zmin), abs(zmax))
        zmin = -zmax
```

Having the colormaps (and symmetric colormap-bounds) as a more convenient option would be very useful indeed.

> 3. Also, there is a managed docked window for displaying multiple plots. Can we have managed docked image windows so that we can display multiple image plots at once?

While this might be possible, I'm afraid it seems to require quite some tinkering with the code it seems. You could have a look at modding the `DockWidget` in `dock_widget.py` (L122 sets which type of widget is used, but that does not seem to be all that needs to be modded). Unfortunately, I don't have the time at this moment to see how to get it working completely.

PS sorry for the slow response. If/when I have time, I will see if I can get a beter awnser

PkRPintu, 2025-08-20:
Thank you Casper and Benedikt,

I have tried to implement the colorbar with the following functionalities.
1. Colorbar appears as a dock widget within the PyMeasure GUI so that one can grab it and put it wherever they want.
2. The colorbar is interactive, which means the higher and lower bounds can be set by moving the sliders inside the colorbar.
3. Few colormaps can be chosen from the Colorbar control widget, which supports both linear and diverging colormaps like viridis, plasma, inforno, magma, cividis, CET-D1, CET-D4,.....
4. For a diverging colormap, there is an option to set Symmetric bounds.

Below, I will share the modified curves.py, the image_gui.py example, and a few screenshots for the colorbar functionalities.

[image]

[image]

[curves.py](https://github.com/user-attachments/files/21899084/curves.py)
[image_gui.py](https://github.com/user-attachments/files/21899083/image_gui.py)

Best wishes
Pradeep