pymeasure/pymeasure #1301: Use dependabot for github actions

BenediktBurger, 2025-05-22:
We could configure dependabot to automatically update github actions (opening PRs, merging is still on us).

What do you think @pymeasure/pymeasure-maintainers ?

In https://github.com/pymeasure/pyleco/pull/120 I added a configuration to update the actions. Also it is necessary to activate dependabot in the settings (Settings -> Advanced security -> Dependabot: enable version updates).

BenediktBurger, 2025-05-23:
Note: An updated ruff action will require a new ruff version with a different configuration (see pyleco)

BenediktBurger, 2025-05-23:
Thanks for your insight and good to hear from you.

Dependabot is (per default) limited to 5 open PRs.
I suggest to use it only for github actions, which we do not have on our map that often.
As they do not change often, I do not expect many PRs, anyways.

For Python dependencies, we want to maintain backward compatibility, so I do not see the use case of dependabot suggesting the most recent version.

bilderbuchi, 2025-05-23:
Sounds good!

CasperSchippers, 2025-05-24:
Sounds good to me indeed!

BenediktBurger, 2025-06-10:
@bilderbuchi could you take a look at the dependabot settings and the PR (each in a subissue)?