pymeasure/pymeasure #1319: Errors when installing pymeasure development version

enlightralucas, 2025-06-16:
Hello !

I want to contribute to pymeasure so I followed the instructions of [Contributing](https://pymeasure.readthedocs.io/en/latest/dev/contribute.html).

I am using a Conda environment with Python 3.12. 

I wanted to test the examples scripts to make sure everything was fine.

However, it seems that I get multiple errors across different files (see attached VSCode picture for errors in gui.py and adapter.py).

[image]

Have I missed something in the development installation procedure ?

Thank you for your support !

BenediktBurger, 2025-06-17:
Pymeasure was written without type hints and some parts do work, but type checkers raise the errors you show here.

Probably you did everything correct. At least, if you have some VISA backend, for example pyvisa-py (or from NI...).

Probable we could rework those parts to proper typing, if we have decided to include typing.

enlightralucas, 2025-06-17:
Thanks for the clarification!

The example scripts work as expected, so everything seems to be set up correctly on my end.

Also, great to see that you opened an issue for adding proper typing — that would definitely be a big help for developers like me working with static analysis tools. Appreciate the support!

closed 2025-06-17 by enlightralucas