pymeasure/pymeasure #1320: Type hints for pymeasure

BenediktBurger, 2025-06-17:
What do we want to do about typing, @pymeasure/pymeasure-maintainers, @cjermain  ?

### Options

1. Basically no type hints as currently in place
2. Allow type hints, whenever a developer wants to
3. Require type hints for new parts
4. Require type hints for touched parts
5. Rework pymeasure systimatically

### Reasoning

- In #1319 , type errors are a reason for confusion.
- It helps writing tests (autocompletion of instrument properties/methods), see #1084 

### My opinion

I'd start with Option 2 and once some framework is started, we can require it (3 or even 4).

BenediktBurger, 2025-06-17:
As a consequence, we should adapt the documentation.

https://github.com/pymeasure/pymeasure/blob/d859c8d3c2d4ac4364ab527eb8a116d69e27593e/docs/dev/coding_standards.rst?plain=1#L32-L33

bilderbuchi, 2025-06-25:
Regarding #1319, I'm not at all sure if the reported problems are not just a failing of Pylance, at work, I find the quality of Pylance static analysis/linting _quite_ subpar. :-|

PR #1324 (2025-06-30): Update documentation regarding type hints
Closes #1320
closed 2025-06-30 by BenediktBurger